prusa3d/MK3_Power_Panic #1: C5 and C2 are not connected on the positive side on the board layout

bwattendorf, 2017-12-06:
When I run the DRC and show unconnected I get an error that C5 and C2 are not connected. I also get an error that pad 3 on P9 and  pad 1 on P8 are not connected but it looks like they are connected when I manually trace the connection.

ok1hra, 2017-12-06:
Yes you're right, C5 are connected versus the original scheme to the output.
It is does not change detector function.
Thanks for feedback.

bwattendorf, 2017-12-06:
Thanks Dan

Bruce W1BAW

ok1hra, 2017-12-06:
I see happy ham radio operator also in comerce contact.
CUL on bands Bruce
73, Dan OK1HRA

Zdraví
Dan Pech

Prusa Research s.r.o.
Partyzánská 188/7a
170 00, Praha 7

bwattendorf, 2017-12-06:
same here 73

bwattendorf, 2017-12-06:
Closing the issue since it is not a problem

closed 2017-12-06 by bwattendorf

make-ing, 2018-03-21:
Hello, i also came across this error (and already opened an issue #5) and it seems very unlikely that this will work the same as described in the schematic. A capacitor (C5) of this size (10µF) will definitely delay the AC_FAULT signal. I hope you will fix either the error in the schematic or in the PCB since i think this will otherwise lead to some more questions and lot of duplicate issue when other people stumble across this inconsistency.

Thanks a lot for the great work so far!

ok1hra, 2018-03-21:
yes, will be fixed in next version

bwattendorf, 2018-03-26:
Dan do you guys need any Beta testers for the new Multi material upgrade?

ok1hra, 2018-03-27:
probably no, sri

bwattendorf, 2018-03-27:
Thanks for the Reply

boelle, 2018-08-10:
how can i fix this now? i means it must sure be a simple task?