prusa3d/MK3_Power_Panic #2: BOM upload

madhunm, 2018-01-23:
Can you please upload the BOM file; with specific part numbers?

Thanks.

ok1hra, 2018-01-31:
You can generate from KiCad source

closed 2018-01-31 by ok1hra

madhunm, 2018-02-19:
The KiCad source only has values and packages. Not the exact part number.

ok1hra, 2018-02-19:
But C and R these are the ordinary types, and IOs type exactly specified.

boelle, 2018-08-10:
this is what i have picked from mouser

Reference, Value, Footprint, Datasheet, Mouser
"R1","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"R4","33k","Resistors_SMD:R_0805_HandSoldering","","660-SG73S2ATTD3302F"
"R5","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"R2","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"R6","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"P8","AC_FAULT","Connectors_Molex:Molex_KK-6410-02_02x2.54mm_Straight",""
"U1","LTV-354T_","Opto-Devices:Optocoupler_SMD_LTV-817S",""
"R3","10k","Resistors_SMD:R_0805_HandSoldering","","660-RK73H2ATTDD1002F"
"Q2","MMBF170LT1G","TO_SOT_Packages_SMD:SOT-23",""
"R7","100k","Resistors_SMD:R_0805_HandSoldering","","660-RK73H2ATTE1003F"
"Q1","MMBF170LT1G","TO_SOT_Packages_SMD:SOT-23",""
"R8","10k","Resistors_SMD:R_0805_HandSoldering","","660-RK73H2ATTDD1002F"
"D1","D_ALT","Diodes_SMD:D_SOD-323_HandSoldering","","863-BAS16HT1G"
"C4","100n","Capacitors_SMD:C_0805_HandSoldering","","710-885012207098"
"C1","0.33uF","Capacitors_SMD:C_0805_HandSoldering","","77-VJ0805V334ZXACBC"
"P7","24V","Connectors_Molex:Molex_KK-6410-02_02x2.54mm_Straight",""
"C5","10u","Capacitors_Tantalum_SMD:CP_Tantalum_Case-A_EIA-3216-18_Hand","","581-TCJA106M025R0150"
"C2","10u","Capacitors_Tantalum_SMD:CP_Tantalum_Case-A_EIA-3216-18_Hand","","581-TCJA106M025R0150"
"R9","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"R11","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"R10","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"R12","11k","Resistors_SMD:R_1206_HandSoldering","","667-ERA-8AED113V"
"P9","measure","Connectors_Molex:Molex_KK-6410-04_04x2.54mm_Straight",""

i have not  added the optocoupler yet and not the connectors... i will edit this post when i have found everything.... part number from mouser is last on each line.... C2 and C5 are not correct yet