prusa3d/MK3_Power_Panic #4: C2 and C5 capacitor, ceramic or tantalum?

libxmike, 2018-03-20:
on the diagram there is "+" which means it is tantalum (or elko capacitor) but on the PCB there is an ceramic capacitor, what is correct? ceramic or tantalum?

make-ing, 2018-03-21:
I would guess that its a normal ceramic capacitor, as there are only very few 10µF (+25V 1206) tantalum caps on Digikey available. It also doesn't seem to be necessary from a capacity or ESR standpoint to put tantalum caps here. But i'm a little bit confused since there seems to be an error in the PCB reagarding to connection of C5. I opened a separate issue about this #5.

libxmike, 2018-03-21:
yes, it is 1206 ceramic.

ok1hra, 2018-03-21:
yes, all is ceramic

amoose136, 2018-05-29:
Well I'm about to build one with some tantalum caps so I guess it will blow up if it really matters.