prusa3d/MK3_Power_Panic #7: Replace 11k resistor with cap

luxarts, 2018-05-13:
Hello!

I was suprised when I saw the 8 resistors at the AC input. Why not a capacitor? I don't think the phase shift between voltage and current matters.
The total resistance it's 88k. In my country the frequency of the AC it's 50Hz. Doing the math:

Xc = 1/(2*π*freq*c) -> c= 1/(2*π*freq*Xc)

To get 88k: 1/(2*π*50*88k) = 36.17nF aprox.
The close comercial value it's 39nF, getting 81.6k aprox.

Less power disipation, less heat, less footprint.

Regards,
Lucas

ok1hra, 2018-05-13:
Yes, but high voltage capacitor more expensive than resistors.

boelle, 2018-08-10:
i agree with @ok1hra, the power consumption is 150 mW and replacing the resistors with a capacitor to save a little just dont justify the more expensive capacitor

closed 2018-08-23 by luxarts