prusa3d/MK3_Power_Panic #11: Ac power plug

boelle, 2018-08-10:
what make/brand and part number is the ac plug

i have looked arround and not found one that will match the board

boelle, 2018-08-10:
yeah, it would be my next step, 

please note that there is a connection issue between C2 and C5

boelle, 2018-08-10:
oh yes... i dont care much about what socket is used... as long i know which one

boelle, 2018-08-10:
[image]

look at the 2 white lines.... it has been discussed on one of the other issues (might be one of the closed ones) all we got to know is that it will be fixed in next version but not how we could fix it our selfes

boelle, 2018-08-12:
@amoose136 like this?

[image]

[image]

amoose136, 2018-08-12:
That's almost what I'm going to do except the board will stick out in the opposite direction because of the a particulars of how my power supply is mounted relative to the switch.

amoose136, 2018-08-12:
[image]
Remap like so, rerun DRC -> refill zones. Warnings go away and connectivity matches the schematic. Thing is this is electrically different from the board as given. In the schematic C5 is on a separate net from P9 pin 3. On the schematic it seems that P9 exists to create probe points for diagnostic reasons but the printer doesn't care about all that. I think if you don't care if ground is on the edge pin of p9, you could swap pin 4 with pin 3 on the schematic and then all of the ground plane can be contiguous on the front so the back would be redundant and the two vias can be eliminated.

amoose136, 2018-08-12:
[image]
[image]
Examples of this change are attached.

boelle, 2018-08-12:
Thanks, would be nice if  @ok1hra  can confirm this is how he would fix it for the next version

Deneteus, 2018-10-08:
The closest I found to original is this Cat. No 57411. Except the Prusa Mk.3 one doesn't have the locking tabs.
http://www.internationalconfig.com/documents/57411.pdf

closed 2019-03-19 by boelle