prusa3d/MK3_Power_Panic #12: Update the design files to reflect the Power Panic integration into the DELTA PSU

CacikeOtoao, 2019-03-04:
@ok1hra , Please update this repo to include the new design of the integrated Power Panic board into the PSU. Even if it is a customized PSU, the design files should be updated since the design changes would need to comply with the Open Source nature of the repo.

maximlevitsky, 2019-03-20:
I second that.

CacikeOtoao, 2019-04-29:
@ok1hra, @josefprusa, can we get an official word on this please?

CacikeOtoao, 2019-07-16:
No design files will be made available. I was told that Prusa Research instructed Delta not to disclose the files nor make the PSU available to third parties.

CacikeOtoao, 2019-07-27:
Since there has been no comment or statement made in regards to this issue I will be flagging this on Github as a GPL violation. This project was obviously modified to be fitted inside a Delta PSU and you did not update the project to reflect that.

maximlevitsky, 2019-07-29:
To be really honest, despite the fact that it is sad to see the powerpanic go (it actually called [UVLO](https://en.wikipedia.org/wiki/Undervoltage-lockout) in technical terms, which I found kind of cool), 
the power panic circuit is just a bunch of resistors to reduce the incoming voltage, then a optoisolator for safety, and then two transistors that do the very basic job of rectification of the current.

In a PSU, the same functionality can probably be done by connecting to some internal line.
Basically a line with low enough voltage but before the big capacitor, so that line looses voltage immediately and without delay.

So I am inclined to think that there is nothing from the actual GPL code that was used.

Deneteus, 2019-08-12:
Nothing in this design is a secret. Even if he made it so Delta couldn't release the design files it takes maybe 10 minutes (with experience) to open one up and figure it out yourself if you already have one of the new PSU's. I still would want the design out there just in case there is an issue with their PSU that nobody knows about. There have been too many safety issues with PSU's failing due to bad designs to not want to know how it was integrated.
https://www.edn.com/design/test-and-measurement/4361392/Circuit-monitors-ac-power-loss

ok1hra, 2019-08-19:
Now available here https://github.com/prusa3d/MK3_Power_Panic/blob/master/rev._Delta_PMZ-24V240WDAL/Delta_PMZ-24V240WDAL.pdf

closed 2019-08-19 by ok1hra

maximlevitsky, 2019-08-21:
I am proud of the decision to release this.
Even though this is mostly a formal thing since as I explained above, there is not much to hide there anyway, but still.

Happy printing!